AutoMecUA/AutoMec-AD #6: Comments upon file how_to_create_csv.txt from robot_core/src/ML_approach/

Almeida-a, 2021-03-06:


Almeida-a, 2021-03-06:
Should be explicitlty stated in what path the commands should be run or what difference does it make.

manuelgitgomes, 2021-03-06:
I believe line 23 on the txt is outdated, referencing a line that does not say anything near the replacement suggested. Is it line 280 now?

manuelgitgomes, 2021-03-06:
@DanielCoelho112

DanielCoelho112, 2021-03-06:
Yes, there was a change, the correct line is 342.
And the content has also changed:
<origin xyz = "0.3 0.011 0.45" rpy = "0 0.6 0" />
But I will also change the txt file

miguelriemoliveira, 2021-03-06:
Hi @DanielCoelho112 and @manuelgitgomes ,

I am receiving notifications for these issues and read through your conversation.

Just a tip: if you want to refer some code lines in a file which is in the repository, go to the file, click of the line number and select **copy permalink**. You get something like this, which is better than your way of referring to code:

https://github.com/DanielCoelho112/AutoMec-AD/blob/efdb51e58ae9913792f016d1746bc17d119287c9/core/CMakeLists.txt#L18-L19

Here's an example of an issue I am working on:

https://github.com/lardemua/atom/issues/251

Regards,
Miguel

manuelgitgomes, 2021-03-06:
Thank you for the tip, professor!

manuelgitgomes, 2021-03-07:
On write_csv.py, my code failed, however, when I changed line 1 from `#!/usr/bin/env python` to `#!/usr/bin/env python3` it fixed. Does anyone have a clue on why that is?

miguelriemoliveira, 2021-03-07:
Hello,

sorry to meddle in the discussion once again.

That is called a shebang line. It tell the terminal which language to use
when running the code in the file.

Since python2 and python3 exist simultaneous in some OS, it is possible to
use on or the other by running:

#!/usr/bin/env python2

or

#!/usr/bin/env python3

respectively. If you want to run the python version which is the default on
your computer you use:

#!/usr/bin/env python

which will call whatever your OS has defined as default. You can find the
default using

python --version

Newer versions will have python3 as default since python2 has support
terminated as of january 2021.

Hope that helps,
Miguel

DanielCoelho112, 2021-03-07:
@manuelgitgomes  Just let me make a change to the camera parameters and then run write_csv.py again

DanielCoelho112, 2021-03-07:
Okay, now you can run the file @manuelgitgomes 
If you continue to have problems on the shebang line, tell me that we will solve it.
Thanks for the explanations professor! @miguelriemoliveira

manuelgitgomes, 2021-03-07:
Thank you @miguelriemoliveira for the explanation!
Ok, I will try it now!

closed 2021-04-03 by DanielCoelho112