AutoMecUA/AutoMec-AD #23: Build from scratch a CNN (Convolution Neural Network) multi output regression that accepts gray images as inputs, and returns both the steering angle and the velocity

DanielCoelho112, 2021-07-03:


DanielCoelho112, 2021-07-03:
Not so good results... 

https://streamable.com/30nlrl

The model with constant velocity is much better, and this one took almost 1h in training, however, it is interesting to watch the robot slow down and search for the best position. 

For the training, I used 2 laps, but I have to admit, controlling the steering angle and the velocity is not an easy task...

manuelgitgomes, 2021-07-04:
Very interesting results, but with that training time, it does not seem that promising

DanielCoelho112, 2021-07-04:
True, but I increased the frequency up to 25 Hz, (2.5x) and used 40 epochs (2x) so I can't tell you if this new CNN is slower than the other or not.

manuelgitgomes, 2022-05-16:
Possibly not feasible, maybe use RNN or reinforcement learning/deep reinforcement learning,

closed 2022-05-16 by manuelgitgomes