AutoMecUA/AutoMec-AD #25: cnn1 train crash with RuntimeError: main thread is not in main loop

inaciose, 2021-08-10:
Exception ignored in: <function Image.__del__ at 0x7fae27637d30>
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 4017, in __del__
    self.tk.call('image', 'delete', self.name)
RuntimeError: main thread is not in main loop
Exception ignored in: <function Image.__del__ at 0x7fae27637d30>
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 4017, in __del__
    self.tk.call('image', 'delete', self.name)
RuntimeError: main thread is not in main loop
Tcl_AsyncDelete: async handler deleted by the wrong thread

inaciose, 2021-08-10:
Solved. Solution. Change code in file:
cnn1/utils.py

#import matplotlib.pyplot as plt
import matplotlib
matplotlib.use('Agg')
from matplotlib import pyplot as plt

closed 2021-08-10 by inaciose

miguelriemoliveira, 2021-08-10:
Hi @inaciose ,

very happy to see you on Automec!