AutoMecUA/AutoMec-AD #27: CNN2 Problem with big datasets

DanielCoelho112, 2021-08-10:
Arrays sizes exceed available RAM.
Possible Solution: Transfer Learning, this way, we would train the same model multiple times but using different datasets

DanielCoelho112, 2021-08-10:
Problem solved.
I implemented the transfer learning in the script "train2a_transfer_learning.py".
Please, somebody test to see if works on any PC besides mine. The var "n_times_tf" represents the number of times that the training will occur, currently is 10, but is supposed to change depending on the dataset used.

manuelgitgomes, 2021-08-24:
I tried to run it. It gave me an odd output. See if I have done someting wrong. I used the training2a_tf.launch to run it. It gave me this output: 
[image]

DanielCoelho112, 2021-08-31:
@manuelgitgomes usually, this odd output occurs when you forget to add the extension .h5 to the model name. Can you try again?

manuelgitgomes, 2021-09-09:
Found a bug, might not be very useful, but when number of right/left angles are 0, code breaks due to dividing by 0
[image]

manuelgitgomes, 2021-09-09:
Nevertheless, it runs well!
[image]

manuelgitgomes, 2021-09-27:
Shouldn't this be marked as closed?

DanielCoelho112, 2021-09-30:
I believe it is done, however, please review the pull request and then merge.

manuelgitgomes, 2021-10-02:
Another thing we need to add is a launch file for this cnn on physical_bringup and simulation_bringup. I will take care of this now.

DanielCoelho112, 2021-10-02:
That is already done. We only need a new launch for training (and that is already done in `traxxas_training2a_tf.launch` ) because, for the driving part, we should use the `traxxas_drive2a.launch`.

manuelgitgomes, 2021-10-02:
Sorry, my bad. But I believe it is only done on simulation_bringup and not on physical_bringup. I know they are interchangeable, but for consistency's sake, I think they should be on both packages. Do you agree?

DanielCoelho112, 2021-10-02:
Yes, I didn't notice that. I forgot to add the launch in physical_bringup. So please add the launch on physical_bringup and then I believe we can close this issue.

manuelgitgomes, 2021-10-02:
Done on #69