AutoMecUA/AutoMec-AD #31: The camera2topic.py did not work well after the first launch

inaciose, 2021-08-11:
The camera2topic.py gives an error about empty numpy on relaunch.

The camera2topic should be replaced by the use the usb_cam package that gives better functionalities including camera calibration.

You need to install the dependencies first.
sudo apt install ros-noetic-usb-cam

DanielCoelho112, 2021-08-11:
That error is because the index camera is wrong. Try another one other than 0.
https://github.com/AutomecUA/AutoMec-AD/blob/591fa57fc98d8d03f1945507e62b6b2ba9c36da0/robot_driving/scripts/camera2topic.py#L15-L16

inaciose, 2021-08-11:
> That error is because the index camera is wrong. Try another one other than 0.
> https://github.com/AutomecUA/AutoMec-AD/blob/591fa57fc98d8d03f1945507e62b6b2ba9c36da0/robot_driving/scripts/camera2topic.py#L15-L16

That what I supposed, the default its 2, and when it stop to work I tried all the video devices listed in /dev, and it didnot work in none of them,  but usb_cam works in the 2, if i quit usb_cam node and try again with camera2topic it still gives the error, launch usb_cam it works. Allways whit the same device.

http://wiki.ros.org/usb_cam

manuelgitgomes, 2021-08-24:
That is weird, I tried it now and the error did not repeat. In which launch file did that happen?

brunomonteiro98, 2021-08-25:
The error today in the physical car was because of that, the camera id . I the drive_write.launch the camera id was set = "0" and in all the others was set = "2" , the 2 being the correct one