AutoMecUA/AutoMec-AD #32: Has anyone managed to make a complete lap on the gazebo with the ackermann robot?

DanielCoelho112, 2021-08-15:
I've tried with all our CNN models (cnn1, cnn2a, cnn2a_tf) and they all have some kind of problem.
I don't know if the issue is my training driving or if the problem lies in camera height and camera inclination...

Can any of you see if you have better results? If not, I think that the best approach is to change the camera position.

inaciose, 2021-08-16:
I also notice that. I never changed the height and camera inclination, and it drive well at the begin. As far as I know the only thing that change in the model was the image size from 200 x 60 to 320 x 160.

manuelgitgomes, 2021-08-17:
I have not, I can try it later. I will bring results at the end of the day

manuelgitgomes, 2021-08-17:
In my limited training in cnn, it ran quite well on a few laps, with only one major hiccup. Nevertheless, we should investigate this issue further.

DanielCoelho112, 2021-08-18:
Which CNN did you used?  @manuelgitgomes

manuelgitgomes, 2021-08-19:
CNN1 @DanielCoelho112

brunomonteiro98, 2021-08-19:
Also used cnn1 with 3 laps of training , 0.5 m/s and 20 epochs and managed to make the first corner but it didn't exit to the crosswalk section. When in autonomous mode the robot view seemed a bit low . Also another thing, in my experience the way we drive, with a limited camera view in our case, has a significant impact on the results, example: when exiting the first corner it is best to stay on the right side , that way the car as always a continous line to follow

manuelgitgomes, 2021-08-19:
A quick note, my PC is very slow and runs gazebo at a real time factor of 0.2 or less. So it means that I was recording a 5 times larger dataset then yours.

joaomecreis, 2021-11-06:
@inaciose with kinect model the image topic comes with delay and does not save the images with traxxas_drive_write. Can you help me?

brunomonteiro98, 2021-12-28:
In these 2 weeks we should focus on gazebo

manuelgitgomes, 2021-12-29:
@Ze6000, you need to create an URDF file for each new angle and/or height, using the format robot05_<angle>_<height> for the file name.

Ze6000, 2021-12-29:
After changing the rotation of the camera to 0.4, the robot was able to conclude 5 complete laps in a row
The data was around 8000 images and use the cnn1 training with 100 epochs of 50 repetitions

DanielCoelho112, 2021-12-29:
Hi @Ze6000,
Great job! What do you mean by "repetitions"?

Following what @brunomonteiro98  said, I think we should collect data from different scenarios and then train a cnn with all data. By different scenarios I mean: different illumination, images with random noise, etc.

Ze6000, 2021-12-29:
Each epoch has a number of "steps" until it's complete. @manuelgitgomes does 100, I use the default, 50

Ze6000, 2021-12-30:
After a few tests, cnn1 runs great with 0.38 camera angle.
I record a data set and trained a model, everything is on teams so you can record more data with other scenarios.

closed 2022-01-04 by manuelgitgomes