AutoMecUA/AutoMec-AD #33: Create a 0-10 scale to rate each machine learning model

DanielCoelho112, 2021-08-15:
In order to rate each created model, I think it would be useful to mark each model with a classification (0-10), and that classification could be stored in a yaml file with the same name as the model. Alongside the classification, others properties can also be stored, such as, developer, image_size, frequency, linear velocity, environment (gazebo or real), camera_position, and cnn_number.

The classification can be something like:

0- Looks random behavior
1- ...
10- Drives perfectly

Any suggestions to complete the classification rate and the properties to store in the yaml?

manuelgitgomes, 2021-08-17:
That seems like a very good idea. We had already talked on a similar feature, so we can drive with the same image dimensions that the model had been trained with.

brunomonteiro98, 2021-08-19:
I suggest simplyfing to somethig like : 0- Looks random behavior 1: doesn't drive well  2: drives okay 3: almost perfect 4: no errors noticed. In terms of storing the gazebo or real models we can work on diferent directories and identify that way, but we can still put in the yaml file.  The rest of the stuff sounds fine by me

manuelgitgomes, 2021-08-19:
What are the benefits of using yaml instead txt?

DanielCoelho112, 2021-08-20:
Agree with your rating scale @brunomonteiro98 . For me, it's set!

DanielCoelho112, 2021-08-20:
@manuelgitgomes I think we should store the metadata in yaml files for two reasons:

1. Using a yaml file, we can normalize the file structure through key-value pair, which makes the file more elegant, see figure.
[image]
[](url)
2. Using a yaml file we can load the file as a dictionary, using python, and perform some data analysis if necessary.

Do you agree? Or do you prefer to use another type of file?

DanielCoelho112, 2021-09-02:
After the weekly meeting we concluded that the yaml is indeed the best option and we should add 3 more parameters:
-  Number of epochs
-  Number of "transfer learning" (in case of cnn2a_tf)
- Camera inclination

Also, we should be precise concerning camera_position and avoid abstract statements like "regular position".

closed 2021-09-02 by DanielCoelho112