AutoMecUA/AutoMec-AD #50: Explore some IPM code (wip)

inaciose, 2021-09-23:


DanielCoelho112, 2021-09-25:
Implemented the first attempt of inverse perspective mapping in :

https://github.com/AutomecUA/AutoMec-AD/blob/33df772b5a286baccbf0bf6fdfc9212881e15b3c/inverse_perspective_mapping/ipm3.py

I followed the paper https://doi.org/10.1016/j.inffus.2014.09.003 made by prof. @miguelriemoliveira but the results are still quite bad.

[image]

However, I believe this is some simple mistake in my code or simply a wrong parameter, I will come back here when I have some updates.  What about you, @inaciose, have you made any progress regarding this issue?

DanielCoelho112, 2021-09-25:
I found the bug, I was applying the rotation angle in the wrong direction (now it seem obvious looking at the previous image).

[image]

Now I think we have 2 problems :

1.  Increase the code efficiency (it took 4 seconds to perform the ipm)
2.  Find a way to determine the intrinsic parameters of the real camera

inaciose, 2021-09-26:
> I found the bug, I was applying the rotation angle in the wrong direction (now it seem obvious looking at the previous image).
>
> [image]
>
> Now I think we have 2 problems :
>
> 1. Increase the code [...]

1) I notice the bad performance of code, it cannot be used in a RT problem like the one we are dealing.

closed 2021-09-26 by inaciose

inaciose, 2021-09-26:
1) 
662 / 5000
Resultados da tradução
I noticed the performance problem that the code has. As it is, it is impossible to apply a real-time solution.
I haven't had time to take a good look at the code yet, but from what I've seen it can be improved.

Lines 21 to 25 may disappear...
The code that performs preparatory calculations can only be run at the beginning of the node, and each iteration can just run the calculate_new_points_list method.

Another possible solution is to use the node I put there, ipm2.py, which can be executed in real time, and whose difficulty is to find the coordinates of the 4 points in the new plane. using the ipm3 code to calculate these coordinates giving the 4 points. 

2) Essa parte sei fazer, alias ontem passei um bocado a tarde a tentar fazer isso no gazebo.

https://www.youtube.com/watch?v=kmVLs8fAMkU

DanielCoelho112, 2021-09-26:
1. Don't worry about the first problem, there are several steps that can be eliminated, this version is just like that to see if it really works. Also, at the moment the ipm only works with gray images, something that also needs to be changed. And yes, as you said, the 4 seconds refers to the total time, in real time you only need to run one method.

2. Good! So you can open a issue to do that, maybe we can do it on Tuesday?  Besides that, we need to start working on measuring the height and angle of the camera in real life (with precision)

inaciose, 2021-09-28:
This issue should not be closed. its work in progress.
On ipm3.py the coordinates are x,y, but be aware that x is row, and y is column.