AutoMecUA/AutoMec-AD #67: IPM optimization - Implementation of IPM with warp perspective

DanielCoelho112, 2021-09-30:


inaciose, 2021-09-30:
Working cpp ROS node
commit 8be36d
Create ipm1.cpp
ROS node for IPM with parameters defined by slider

inaciose, 2021-10-01:
Working python script (not a ros node) based on ipm1.cpp uploaded
Its fast

DanielCoelho112, 2021-10-02:
Good job! Now, I suggest we clean up the code of that script and leave just the IPM class. This way we can import the IPM class into other scripts to keep the code simple.

inaciose, 2021-10-03:
The cleanup is done. the class can be imported and used in another scripts.
some undistort methods was also added to the class.
with undistort added, each frame takes 3.5 ms to process

inaciose, 2021-10-11:
The today meeting point us some directions. We may proceed to lane detection task using the opencv wrapPerspective to get birds eye view, and working to get better results from the code o ipm_optimized2.py done. 
@brunomonteiro98 cameout with this work (https://github.com/robin-karlsson0/perspective_mapping/blob/master/direct_and_inverse_perspective_mapping.ipynb)

We need to test if the wrapPerspective method, allow to get pretty good estimates of distances on physical world 
I did some rudimentar preliminary tests in gazebo, and it sounds promising.

We may assume that the bev, is ok and tray to develop some lane tracking system. That system only need the bev image to work with it.

We also discute how to improve the results of the simple code and naive code that can be see in action in the next video. 

https://www.youtube.com/watch?v=Rj3GNvw-yBc

@Almeida-a will post more information about how to do the next developments

closed 2021-10-12 by DanielCoelho112