AutoMecUA/AutoMec-AD #71: Adapt IPM class to accept the intrinsic and extrinsic parameters as yaml files

DanielCoelho112, 2021-10-02:
If the parameters are delivered in yaml files, we can easily use the class for simulation cases and physical cases. The only difference in simulation and physical would be the loaded yaml.

inaciose, 2021-10-03:
@DanielCoelho112 
Please see the config structure of the new ipm.py that i upload today
The ros calibration process create yaml files that can be used for intrinsic parameters.
The ipm class should not be changed to load the parameters from yaml files, as it gets the config when the ipm object is created.
The loading should be done by the the script that use the ipm class, maybe by making another class for it.

sample folder structure

app/lib/ipm.py
app/lib/cfgyml.py
app/script.py

script.py
from lib.ipm import IPM
from lib.cfgyml import CFGYML

def main():
  md, mi, me = CFGYML(filename)
  ipm = IPM(md, mi, me)

manuelgitgomes, 2022-05-16:
This does not make sense anymore due to the possibility of retrieving intrinsic parameters from camera_info and extrinsic from URDF.

closed 2022-05-16 by manuelgitgomes