AutoMecUA/AutoMec-AD #78: Project and print 3DP support for PC

manuelgitgomes, 2021-10-07:
As it currently is, PCs are very unstable when place on top of our robot, so there is a need for a support.

brunomonteiro98, 2021-12-27:
@Ze6000  please put the stl files in teams to know what items to print

brunomonteiro98, 2021-12-28:
The version of the gcode with the adhesion option doesn't work well, we need to do an updated version without that

manuelgitgomes, 2022-01-04:
Print on my printer

manuelgitgomes, 2022-05-16:
This needs to be done eventually, we should check out how Atlas does it.

PedroMS3, 2022-07-12:
a prototype has been made, but after further analysis with the rest od the team, it was concluded that we should keep it simple, using some pieces of acrylic, sponge and bolts. Nonetheless, here are some some pictures of the prototype.
[image]
[image]

PedroMS3, 2022-09-28:
we came to the conclusion that it would be better to have a wider board, and also maybe a bit longer to better accomodate the pc

PedroMS3, 2022-10-14:
i got the new mdf board and will bring it to aveiro this week, also i drew the new base in solidworks just have to double check some measures to make the final drawing

closed 2023-10-04 by Ze6000