AutoMecUA/AutoMec-AD #84: Next steps: Algorithm ideas for building a road elements awareness

Almeida-a, 2021-10-12:
This issue is to take as base the birds' eye view image projection, however, it still is useful even if _bev_ is deprecated. Also, this issue serves as starting ground for new ideas for the classical/glass box vision programming development.
Some opening ideas discussed in 11 October (yesterday):

1. Try using the camera from a higher distance from the ground, thus capturing more of the road, prior to the ipm process.

2. Differentiate road lines in two categories: type (continued/dashed), or position (to the left / right / at the front).

3. Use two hysteresis lists for detected lines:
3.1. First has only one element -> compare with current line and accept it only if it does not deviate too much from the past one
3.2. Second has n elements -> contains the two most common line slopes, where there would be two common situations:
3.2.1. The most common directions are the straight ones, this is the usual case.
3.2.2. There is a second emerging direction, along with the straight ones. This is the crossroads case, where the car is likely to turn.

4. Try to obtain real distances, allowing some sense of the car position in the road.

5. Following the topic 4, being aware of the distance (in the horizontal axis) between the road lines. This is a fresh idea, but may open doors to detecting the zebra crossing.

DanielCoelho112, 2021-11-04:
Hi @inaciose, @andrefdre, and @manuelgitgomes ,

Sorry for the absence in the last few days... So, any news regarding IPM and later processing?
About that missing transformation that we talked about a couple of weeks ago, I've looked into it, and I think that's not the problem...

I saw the working plan for the next year posted by @manuelgitgomes and I think that we should integrate the development of IPM also in D1. Although the results of CNN model (end-to-end) are promising I think that I'm not confident enough to bet everything in this approach...  I suggest we work in both approaches independently, and later on, we decide which one we choose for D1, what do you say?

manuelgitgomes, 2021-11-06:
It makes a lot of sense, I will add it to the plan!

manuelgitgomes, 2021-11-18:
The sliding windows method (#104) appears to be a common option to proceed. Whilst IPM code is not completely functional, @andrefdre should work with a ROS model with the camera facing down, similar to how to final product of IPM is.

manuelgitgomes, 2022-05-16:
Due to the current state of ML Algorithms, this is not needed.

closed 2022-05-16 by manuelgitgomes