AutoMecUA/AutoMec-AD #100: Signal holder structure

brunomonteiro98, 2021-11-02:
We need to construct a (wooden maybe) structure to hold a phone to show the signals. 
Check with LAR if they already have something like this

manuelgitgomes, 2021-11-23:
Professor Artur has a working structure, we shall meet with him to learn how to mount it.

manuelgitgomes, 2021-11-30:
Send an email to professor Artur to schedule a meeting

manuelgitgomes, 2021-12-15:
We already have the structure! Now we need to acquire two old monitors!
[image]

miguelriemoliveira, 2021-12-15:
I think you could speak to Eng. Rui Heitor in LAR.

Perhaps he has some.

Regards,
Miguel
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

manuelgitgomes, 2021-12-15:
Thank you professor!

brunomonteiro98, 2021-12-27:
@manuelgitgomes  do we still need the "chaves sextavadas" ?

brunomonteiro98, 2021-12-27:
I will bring the tools

Ze6000, 2022-01-25:
The signal holder structure is amost ready. The only thingh that's missing is a splitter e cables to connects a single pc to both monitors

closed 2022-03-07 by manuelgitgomes