AutoMecUA/AutoMec-AD #103: Update Guia_software document

brunomonteiro98, 2021-11-10:
Update with more ROS content and substituting the obsolete course

brunomonteiro98, 2021-11-23:
We need to schedule a meeting to discuss an updated version of our general Guia_software document. With @DanielCoelho112 @callmesora @manuelgitgomes  and myself.

brnaguiar, 2022-01-11:
Useful links for Computer Vision, Machine Learning and Deep Learning:

First Principles of Computer Vision: 
https://www.youtube.com/channel/UCf0WB91t8Ky6AuYcQV0CcLw/playlists

Deep Learning NYU Course by Yann Lecun and Alfredo Canziani: 
https://atcold.github.io/NYU-DLSP21/
https://github.com/Atcold/NYU-DLSP21
https://www.youtube.com/playlist?list=PLLHTzKZzVU9e6xUfG10TkTWApKSZCzuBI

ML theory:
https://www.youtube.com/playlist?list=PLLHTzKZzVU9e6xUfG10TkTWApKSZCzuBI

Robotics in general:
 https://www.youtube.com/c/CyrillStachniss/playlists

brunomonteiro98, 2022-01-18:
New organizatio: pre requisites + 3 areas: ML / Visao / ROS

ROS: ros wiki

Vision (theoretical) : https://www.youtube.com/channel/UCf0WB91t8Ky6AuYcQV0CcLw/playlists + open cv official wiki https://docs.opencv.org/4.x/d9/df8/tutorial_root.html 

ml : https://www.coursera.org/learn/machine-learning?action=enroll + teoria introdutoria : https://www.youtube.com/playlist?list=PLZbbT5o_s2xq7LwI2y8_QtvuXZedL6tQU + teoria avaçada https://www.youtube.com/playlist?list=PLLHTzKZzVU9e6xUfG10TkTWApKSZCzuBI

manuelgitgomes, 2022-01-25:
@brunomonteiro98 How is the state of this issue?

brunomonteiro98, 2022-03-07:
The new version is already in teams but i need to add a new document with all the relavant links. This is probably better done with a bookmarks page in the wiki.

manuelgitgomes, 2022-03-22:
@brunomonteiro98 Write the text and @Almeida-a formats it.

manuelgitgomes, 2022-05-16:
Closing this issue due to the need of a wiki restructure

closed 2022-05-16 by manuelgitgomes