AutoMecUA/AutoMec-AD #104: Work over sliding windows method

Almeida-a, 2021-11-17:
Sliding windows is a common method to detect lanes in the context of autonomous driving. It was previously used with no meaningful results due to inexperience to further analyze said method.

In this issue we will revive this approach and discuss its potential to solve or help our lane detection objectives. 
This is a continuation to issue #91, where we concluded that this is a viable option.

Almeida-a, 2021-11-17:
The [code](https://github.com/AutoMecUA/AutoMec-AD/blob/new/sliding_windows/lane_detection/script/sliding_windows.py) is in branch `new/sliding_windows`, and its path is `lane_detection/scripts/sliding_windows.py`.

Next steps:
- Test the code's robustness (find more bugs or problems with it).
- Resolve the bug where **one of the road lines is not visible**.
- Automate the gazebo/script running process with roslaunch.

Almeida-a, 2021-11-17:
For good practice, we can use [rospy logging](http://wiki.ros.org/rospy/Overview/Logging). Here for a tutorial [page](http://wiki.ros.org/rospy_tutorials/Tutorials/Logging) over it.

andrefdre, 2021-12-01:
I created a new model witth the camera view from the top. I'm remaking the code.

andrefdre, 2021-12-14:
I based the code in a github page and was able to start detecting lines. Right now its onlt able to detect 2 lines at a time.
One of the aproachs is to change the code to detect 3 lines instead of 2. Another aproach is to remove the middle parte of the image so the middle lane isn't detected until one of the side lanes stops being detected and then uses the midle lane too.

andrefdre, 2022-01-12:
Added the part of the code that calculates the position from the center and the curvatures of the road.
Added also the detection of lines using the previous image information obtained from the histograms

manuelgitgomes, 2022-05-16:
Due to the current state of ML Algorithms, this is not needed.

closed 2022-05-16 by manuelgitgomes