AutoMecUA/AutoMec-AD #105: IPM Optimization

DanielCoelho112, 2021-11-18:
There are several ways that we can optimize the current IPM optimization, such as:

- Decrease code execution time
 Use matrix multiplication instead of loops
- Increase output quality image
 Verify intrinsic matrix
 Verify extrinsic matrix
 Look for bugs in the code

DanielCoelho112, 2021-11-18:
The current IPM approach returns an image like this:

[image]

As you can see, the results are not optimal... And it took about 0.4 seconds to process each image...

manuelgitgomes, 2021-12-04:
IPM package fully operational, although it seems results are worse. Further work here is needed.
[image]

manuelgitgomes, 2021-12-04:
The IPM class now receives a dictionary with the camera pose and calculates the extrinsic matrix from it. The intrinsic matrix is retrieved from a ROS topic published by our vehicle. The results stay the same as they were before my last comment, so they are the same as @DanielCoelho112 described. Now, sliders will be added to the image in order to change values and certify our calculations are correct.

Almeida-a, 2021-12-06:
As the commit complementary message describes (click the "..." button on that commit message), this slider implementation still has some problems.
They will be addressed perhaps at our next meeting, and the appropriate changes will be carried out to fix them.

PS: I also added a TODO and a FIXME in code to pinpoint those problems.

Almeida-a, 2021-12-07:
The remaining code sections to be reviewed for optimization are the following functions:

- calculate_A_const
- calculate_extrinsic_matrix

Almeida-a, 2021-12-11:
I found an interesting concept which we can use to carry out our optimization, which is profilers.
Here is a short [video](https://youtu.be/QSueV8MYtlw) on it, good for introducing that concept using pycharm.

Almeida-a, 2021-12-11:
This [post](http://stackoverflow.com/questions/29433788/ddg#29435952) might fix our problem that is causing the slide bars not working.

manuelgitgomes, 2021-12-11:
Regarding the sliders, this week I have implemented sliders on the signal recognition, as you can see here. You can see it and use it as a base for tour sliders!
https://github.com/AutoMecUA/AutoMec-AD/blob/af9cfbaa1e81c8af7e70101f7ada3dc392c0cdbe/signal_recognition/scripts/color_segmenter.py#L21-L50

manuelgitgomes, 2021-12-20:
I have tried a different approach with better results than before. It's a mix between the one suggested by @inaciose and the one done by @DanielCoelho112. Instead of transforming all the image into the new coordinates, I chose only four points on the image to convert them. After finding the coordinates, use the `cv2.getperspectivetransform` function to create the output image. The process happens almost instantenously. Here you can see the selection of the corners and the resulting image.
[image]

[image]

Some things I yet cannot explain. The roll variable and the pitch variable seem to be switched. And the transform is not yet perfect. Feel free to comment.

manuelgitgomes, 2022-03-15:
Retry rotation matrix calculation using
 https://github.com/GabrieleMicheli/PSR_A3/blob/0db343986b5595e496ab718c6abd6acc70b02232/p_group8_player/src/driver.py#L779-L791

manuelgitgomes, 2022-03-21:
[image]
As seen, IPM is now much better. It does not, however, show the whole image. This might have to do with the parameter dsize in cv2.warpPerspective. 
Further work needs to be put here.

manuelgitgomes, 2022-03-22:
Issue not a priority for current time.

rospuye, 2022-12-08:
The problem seems to be that the `cv2.getPerspectiveTransform()` function only considers the cropping resulting from the four points in the input image for the output, as this image suggests.

[image]

It also explains why the output is rotated, altough that seems easy to fix. I'm not sure if the cropping problem can be solved through some input parameter of  `cv2.getPerspectiveTransform()` or if we need to use a different function altogether, but I will look into it. I found useful information [here](https://rospibot.azw.pt/inverse-perspective-mapping-em-python-usando-warpperspective/), [here](https://theailearner.com/tag/cv2-getperspectivetransform/) and [here](https://stackoverflow.com/questions/13063201/how-to-show-the-whole-image-when-using-opencv-warpperspective).

manuelgitgomes, 2023-03-13:
[This](https://towardsdatascience.com/a-hands-on-application-of-homography-ipm-18d9e47c152f) approach seems like a valid one. It creates a plane on the floor and projects the points to the pixel image.
The image then receives a bilinear interpolation to tackle the pixel sparsity.

PedroMS3, 2023-06-08:
If the opportunity rises this issue should be opened again, but only if new software members want to work on it

closed 2023-06-08 by PedroMS3