AutoMecUA/AutoMec-AD #106: CNN with difference image

DanielCoelho112, 2021-11-21:
I was researching through some topics about AI, and I found a video of an AI model playing the pong game: https://www.youtube.com/watch?v=YOW8m2YGtRg&list=TLGG2Hi1VyMnZ7gyMTExMjAyMQ&t=2s&ab_channel=AndrejKarpathy

(The person who built this model was Andrej Karpathy, current director of artificial intelligence at Tesla)

The model uses, among other things, a CNN like us. But what really surprised me was the image input of that CNN. Unlike us, to predict the action in time t, the model doesn’t use only the frame at time t, it uses the difference between the frame at time t and the frame at time t-1. I think this could help us improve our driving performance because now the model not only can understand where it is but also understand from where it came from. 

I know that in the previous meeting I suggest that we should focus on IPM and lane detection, but I think that this improvement wouldn’t require major changes in the code, so I think we should check it out. 

What do you think?

manuelgitgomes, 2021-11-21:
It seems really interesting! After the IPM issue is solved (hopefully) this week, we can check it out!

manuelgitgomes, 2022-05-16:
https://github.com/stewy33/pong-with-policy-gradients
@brnaguiar will read further into this to see if it's interesting.

manuelgitgomes, 2022-05-16:
RL implementation will be moved to #124

closed 2022-05-16 by manuelgitgomes