AutoMecUA/AutoMec-AD #107: Alternatives for physical track

manuelgitgomes, 2021-11-23:
Due to some issues, the available track currently held at FCV will be no more. Research and procurement of alternatives are necessary. @PedroMS3 has volunteered to procure plastic sheets (around 40m²) and bring more information.

manuelgitgomes, 2021-11-23:
An email should also be sent by me to CMA and CMI.

PedroMS3, 2021-11-30:
i found that the solution i was looking for was too expensive for our budget, and also reflected a lot of light, which might disturb the camera's ability to perceive the track. I searched at Bricomarché for a kind of plastic wrap, there were 3 sizes 4x4, 4x6 and 4x10, the last 2 were not available.

manuelgitgomes, 2021-12-28:
Two options arose. One near Mercado Municipal de Ílhavo:
[image]
And one near Estádio Municipal de Aveiro:
[image]

manuelgitgomes, 2022-01-25:
The option inside EMA was closed. A  picture:
[image]

To note: In a brighter environment there should be darker lines.
Closing issue,

closed 2022-01-25 by manuelgitgomes