AutoMecUA/AutoMec-AD #109: Global Debug option (for pycharm)

Almeida-a, 2021-12-04:
We have found a way to use pycharm's step-by-step debugging mode when running a python program [here](https://github.com/AutoMecUA/AutoMec-AD/commit/0f417531155ad42caac4127fc0e6330402fc9220), and abstracted the option to the respective launch file [here](https://github.com/AutoMecUA/AutoMec-AD/commit/d14a635307cccd1d0f234589ced0fd551da64b41). After this last commit, one can simply:

- Add configuration to a python debug server (in pycharm, will add it to the wiki), if such is non-existent
- Run that configuration
- Run: `roslaunch ipm ipm_op.launch debug_mode:=True`

And start the debugging at will.

For those who like to use this pycharm feature for debugging, is there a simple way to create an abstraction such that any launch file (that runs a python program) can use that option to the same effect?
Scheme: `roslaunch <any package> <any python launch file> [debug_mode:=True]`.

Almeida-a, 2021-12-04:
Instructions for using the aforementioned debugging feature for pycharm will be futurely added in the wiki.
We may also try to find a way to do the same in VSCode IDE and add the tutorial as well.

Almeida-a, 2021-12-04:
I was thinking and the solution could be the following:
* Creating a "configuration" file where the pycharm settrace function would reside here
* That function would be called only if debug_mode was set to True
* The variable debug_mode would be accessible through `rospy.get_param` (or other)
* Any python file to be executed could simply import the configuration file

Almeida-a, 2021-12-09:
Another solution could be not including a debug_mode variable, and instead try to avoid the ConnectionError-111.
Thus, only by launching the debug server (prior to the main program launch) would the debugging mode be set, otherwise no debug process would be carried out.

manuelgitgomes, 2022-05-16:
Is not useful now. Over engineering.

closed 2022-05-16 by manuelgitgomes