AutoMecUA/AutoMec-AD #112: CNN - Add noise to the steering anlgle in the data collection

DanielCoelho112, 2021-12-15:
I found something that can help us improve our cnn training. 

When we are manually driving the car to collect the data, we always try to keep the car a certain distance from the lane. Although this may seem logical at first, this can prejudice the performance of the model. To keep the car a certain distance from the lane, sometimes we have to think ahead, and this is something that our model can’t do. Furthermore, when we force the car to follow the same path, the first moment the car sees an unknown scenario, it would be completely lost. 

From what I have seen in some papers, some authors produce some kind of random noise in the data collection step and then the driver has to correct the car. For example, every 5 seconds we add to the steering angle a random value between 0 and 1. This will produce datasets with richer information and therefore the driving performance is expected to increase.

brunomonteiro98, 2021-12-15:
I agree with your ideia and i have been doing that when driving the car ( the physical car and in gazebo as well) : instead of just following the imaginary center line I purposely drive with variable offset distance from the center. For example: in the lap 1 I drive with -2X distance from center , lap 2 -X , lap 3 0X , lap 4 +1X and lap 5 +2X . On top of that I introduce "errors" in my driving, for example: trying to go to the limit of the track and then correcting the trajectory (on multiple parts of the track and with multiple inicial offset distances) . I think this aproach achieves what you are talking about without addicional software complexity because i can drive, with around 20 laps in the physical track in DEM, on pretty much all the possible "paths" the car can take, except for paths that take the car *completely* off the track ( which happens sometimes ).

DanielCoelho112, 2021-12-16:
Perfect!

closed 2021-12-16 by DanielCoelho112