AutoMecUA/AutoMec-AD #118: Light sources and noise Gazebo

brunomonteiro98, 2022-01-04:
We need a way to add noise to our world in gazebo to better represente reality.

brunomonteiro98, 2022-01-11:
We weren't able to produce enough "noise" (with adicional light sources) because the floor isn't refletive enough. Maybe Professor @miguelriemoliveira  might helps us with this.

manuelgitgomes, 2022-01-25:
We should check the .gazebo file of our world to see the properties. In last case, an object with infinitely large are and infinitesimally small thickness should be placed between the car and the track. That object would have the reflecting properties

manuelgitgomes, 2022-05-16:
This can be done on data augmentation of cnn, so is not needed.

manuelgitgomes, 2022-05-16:
Possibly add random models to gazebo.

closed 2022-05-16 by manuelgitgomes