AutoMecUA/AutoMec-AD #124: Autonomous Driving Reinforcement Learning

Joaoprcf, 2022-03-01:
We need a reward system given by gazebo services.
The first Reinforcement learning technique applied will be ARS (Augmented Random Search), which will be improved with a 2-3 layer neural network after the initial training (step 2), this new structure will have its weights on the first layers and the last one can be retrained again with ARS. The inputs given to this model can be obtained through CNN preprocessing and others. A script with capabilities to spawn and delete models is required.

manuelgitgomes, 2022-03-22:
Revise requirements.txt.

manuelgitgomes, 2022-05-16:
ROTA implementation should also be studied @brnaguiar

manuelgitgomes, 2022-06-13:
For the referee try researching for frenet coordinates:
https://www.selfdrivingcars360.com/glossary/frenet-coordinates/

manuelgitgomes, 2022-07-04:
Search for transformers use in autonomous driving

manuelgitgomes, 2022-11-28:
Study the code in:
https://github.com/AutoMecUA/AutoMec-AD/blob/fec62272809c74442c5ee7295ec3230bd90147ed/prometheus_driving/scripts/ml_training.ipynb

This is a CNN and we are trying to change it to an RNN.

PedroMS3, 2023-06-08:
For the moment being there is no member capable of working on this issue, therefore if we could get any member  that knows reinforcement learning would be great

closed 2023-06-08 by PedroMS3