AutoMecUA/AutoMec-AD #126: Car safety system

brunomonteiro98, 2022-03-10:
We need to test the equipment that is going to be used in the competition.

For now, because the rule are not out yet, we need more information.

artur@ua.pt and asousa@fe.up.pt are the contacts to talk about this

manuelgitgomes, 2022-03-15:
Prof. Artur response:
Viva!

O fim do semestre foi intenso e só agora estou a ultimar as regras. Espero terminá-las hoje ou amanhã. Não haverá alterações significativas. Apenas a pista encolherá ligeiramente, passando a largura a ser de 1,40m entre meios das linhas delimitadoras. Ou seja, a largura navegável será de 1,45m.

Cumps,
Artur Pereira

manuelgitgomes, 2022-04-19:
Designed box and sketched electrical scheme, need to send an email to prof. Artur for further enlightenment.

manuelgitgomes, 2022-05-09:
Not needed, closing

manuelgitgomes, 2022-05-09:
On the future, have a stopping system with a controller

closed 2022-05-09 by manuelgitgomes