AutoMecUA/AutoMec-AD #131: Code benchmarking

manuelgitgomes, 2022-03-22:
Code is running slow and needs optimization.
Find out where bottlenecks are.
Try time measurements and python profiling.

manuelgitgomes, 2022-03-22:
Python pdb

Almeida-a, 2022-03-23:
I have used the integrated python profiler from pycharm to carry out the following evaluation:
[image]

We can conclude that the matchTemplate function comprises the biggest bottleneck within the standalone script.
A small optimization was carried out: some unneeded templates for the template matching operation were [removed from the list](https://github.com/AutoMecUA/AutoMec-AD/blob/f408547f509db8835c035929c0a477ac21ac1761/signal_recognition/scripts/signal_panel_recognition_standalone.py#L234-L236).

The resulting profiling operation was as follows:
[image]

Which indicates that the overall cpu time consumption was cut by half.

Further work:
* Evaluate the optimization with the original (ROS) script
* If the optimization was not sufficient, continue analyzing and optimizing the code (using the standalone version and profiler first)

Almeida-a, 2022-03-23:
After the optimizations carried out in the latest commit, the code yielded the following profiling results:
[image]

Further work:
 * Same from my last comment
 * Calculate and compare the cycle time from the latest commit to the first one (the one that yielded the first profiling results, which is the code without optimizations)

manuelgitgomes, 2022-04-04:
After testing the code it was found that the optimization, even after the profiling, was not good enough. 
After searching if the openCV function had the option to search with a smaller number of matches, I could not find anything.
A better search for a bottleneck should happen soon.

Almeida-a, 2022-04-11:
We have tested running the full project using 2 other computers, and it seems that it is working better than on the original one. Thus, we are assuming the problem is not with the code itself, and therefore I think this issue can be closed.

manuelgitgomes, 2022-04-11:
Further testing might be needed but we can close this for now.

closed 2022-04-11 by manuelgitgomes