AutoMecUA/AutoMec-AD #133: JPS printer

brunomonteiro98, 2022-04-04:
whe need to substitute the 4 screws to enable the bed printer to go higher

HugoSachim, 2022-04-11:
the problem regarding the limit on the z axis was due to the physical button itself being loose, regarding the quality, some adjustments will have to be made at the slice level

manuelgitgomes, 2022-05-16:
@brunomonteiro98 will test a file

brunomonteiro98, 2022-05-23:
https://uapt33090.sharepoint.com/:w:/s/OP_AutoMec_geral/EV5pSvuEkpdPpr4yvawW0c4BpZZ3H8uOrzn_1IaruziLmw?e=ehq1V4

brunomonteiro98, 2022-05-30:
The problem now seems to be excessive Z height. This, without changing the hardware, is solved by changing the inicial G code, regarding the zero height, in the slicer software, Cura in our case.

brunomonteiro98, 2022-07-11:
Áll seems to be working after the height screws were ajusted again. The bed was leveled and a print test was done succesfully. Futher tests are necessary to validate the precision of the machine

closed 2022-07-11 by brunomonteiro98