AutoMecUA/AutoMec-AD #140: Pan and tilt for kinect

manuelgitgomes, 2022-05-09:
To have a correct angle measurement for the kinect and movement flexibilitty, a pan and tilt unit is needed.

manuelgitgomes, 2022-11-28:
[image]
Example of a (complicated) pan and tilt

FSFToco, 2022-12-16:
[image] basic model of the pan and tilt system

FSFToco, 2023-02-06:
[image]
added a support on the other side of the servo, to offset the weight of the Kinect, it is shaped like a cylinder to freely move as the Kinect base tilts according to the servo

FSFToco, 2023-03-08:
[image]
updated model of the project

FSFToco, 2023-03-08:
[image]
calculations to check if the 9 Kg/cm servo motor that is available would be enough for the tilt system

manuelgitgomes, 2023-03-27:
You can check PTU units!

FSFToco, 2023-04-10:
[image]
adjusted the size for the CAD servo, still have to fix the design of the support

FSFToco, 2023-04-12:
[image]
calculations for the servo motor responsible for the "pan" portion of the system

FSFToco, 2023-04-29:
[image]
latest update of the CAD model, it is entirely supported by the panning servo motor, so the last piece missing would be a support for the servo that can also be used as a connection between the pan and tilt module and Prometheus

FSFToco, 2023-05-17:
[image]
Added a chamfer to the main body upon realizing it would limit the rotation angle otherwise, the chamfer was made so that the hard limit to tilting forward is 20º

andrefdre, 2023-06-28:
I was searching about the kinect and found out the kinect cameras already have the tilt implemented and there is a ROS package to control this `kinect_aux`. So it's only necessary to develop the pan, since the tilt already could be controlled with ROS.

https://github.com/AutoMecUA/AutoMec-AD/assets/58526188/ba473f9d-1dcb-4c71-b31d-5b2533b29aec

PedroMS3, 2023-07-03:
As the kinectic has a pan , the package is called Kinect_Aux, the design will be changed

FSFToco, 2023-07-28:
[image]
As the Kinect can tilt, we need only the functionality of panning, this design is similar to the previous but removes the elements that were used for tilting the kinect.