AutoMecUA/AutoMec-AD #141: Add encoder to motors

manuelgitgomes, 2022-05-09:
To run the car at a fixed velocity, add an encoder

manuelgitgomes, 2022-05-30:
Ino code used in atlascar2:
https://github.com/lardemua/atlascar2/blob/master/atlascar2_description/nodes/Atlascar2_encoder_v2/Atlascar2_encoder_v2.ino

Used encoder:
https://pt.farnell.com/hengstler/ri32-0-1000er-14kb/encoder-rotary/dp/615985

Example of a cheaper one:
https://www.amazon.es/Codificador-Rotatorio-Incremental-Fotoel%C3%A9ctrico-Aceleraci%C3%B3n/dp/B07R8VSMQH/ref=sr_1_8?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=BAGDS9UKTE1S&keywords=encoder+50ppr&qid=1653945205&sprefix=encoder+50ppr%2Caps%2C105&sr=8-8

Ze6000, 2022-07-18:
Some encoders
https://www.amazon.es/-/pt/dp/B015GYY7XU/ref=sr_1_23?keywords=encoder+giratorio&qid=1655123660&sprefix=encoder%2Caps%2C126&sr=8-23
https://www.amazon.es/Codificador-Rotatorio-Incremental-Fotoel%C3%A9ctrico-Aceleraci%C3%B3n/dp/B07R8VSMQH/ref=sr_1_8?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=BAGDS9UKTE1S&keywords=encoder+50ppr&qid=1653945205&sprefix=encoder+50ppr%2Caps%2C105&sr=8-8
https://pt.farnell.com/broadcom-limited/heds-9100-a00/encoder-rotary-500cpr-2ch/dp/1654864
https://pt.farnell.com/broadcom-limited/aedb-9140-a13/encoder-3channel-500cpr-8mm/dp/1161087
https://pt.farnell.com/broadcom-limited/heds-5640-a06/encoder-rotary-500ppr-3ch/dp/1654889

andrefdre, 2022-07-25:
After some research wasn't able to choose a better encoder than @Ze6000 suggested,https://www.amazon.es/-/pt/dp/B015GYY7XU/ref=sr_1_23?keywords=encoder+giratorio&qid=1655123660&sprefix=encoder%2Caps%2C126&sr=8-23.

manuelgitgomes, 2022-08-08:
The encoder:
https://www.amazon.es/-/pt/dp/B015GYY7XU/ref=sr_1_23?keywords=encoder+giratorio&qid=1655123660&sprefix=encoder%2Caps%2C126&sr=8-23.
Has been delivered and work can start on it.

andrefdre, 2022-09-28:
After some research 2 more encoder opportunity appeared that were more compact. There is one which uses magnets and a TTL one. 
The magnetic encoder wasn't showing good values when connected to arduino so we decided to use the TTL one which is also easier to implement in the car.
To add the encoder to the car we though in gluing one part to the rotor and the other to the motor.
We also thought in adding a 3d printed cover to protect the encoder from dust.
[image]
[image]
[image]

andrefdre, 2022-10-26:
The magnetic encoder was added to the motor and it's counting correctly one full rotation. We glued the magnet to the shaft and the encoder was mounted with some thin wire that was soldered to the motor pins.
[image]
[image]
[image]
We dont have the connector that plugs in to the encoder so we need to buy it. The width of the connector is 13mm.
[image]

andrefdre, 2022-10-26:
After some research the Servo library isn't compatible with interrupts. In a post someone suggested to use the PCA9685 which we already have, some research about i2c and interrupts needs to be done to see if one affects the other.
[image]

andrefdre, 2022-11-23:
Now that the new PCB was made we need to see how the wires of the encoder will connect to the PCB:
- Solder directly
- Some kind of connector

manuelgitgomes, 2022-11-28:
Next step: Run both:
https://github.com/AutoMecUA/AutoMec-AD/blob/e1438c8d650cd70405ed8875258501debccdc72b/prometheus_bringup/src/encoder_magnetic.ino
https://github.com/AutoMecUA/AutoMec-AD/blob/e1438c8d650cd70405ed8875258501debccdc72b/prometheus_bringup/src/int_to_movement.ino
In the same ESP, in different cores.

andrefdre, 2022-11-30:
https://embeddedcorner.com/2021/02/03/install-ros-on-your-esp-board-with-arduino-ide/
Connection with esp32 and pc done as seen above.

manuelgitgomes, 2023-01-03:
Design new way to lock the encoder relative to the motor

manuelgitgomes, 2023-01-03:
@PedroMaia21 should take the pcb home to help in code development

andrefdre, 2023-03-30:
At the end of the last meeting, @Manuelgitgomes and I found out that using 2 cores with ESP gave an error using ROS serial. More research is needed to figure out how we could solve this problem.

manuelgitgomes, 2023-04-03:
Maybe we can use this arduino library: https://github.com/sachin0x18/rosserial_esp32

manuelgitgomes, 2023-04-17:
Without using 2 core functionality, we have achieve (briefly) a fully functional encoder/motor control system.

manuelgitgomes, 2023-05-01:
Encoder fully functional

closed 2023-05-01 by manuelgitgomes