AutoMecUA/AutoMec-AD #143: Clean code

manuelgitgomes, 2022-05-09:
Due to last-minute changes and bad coding practices, code cleansing needs to happen.
- [ ] Clean CNN code, implement naming for datasets and models - @brnaguiar;
- [ ] Clean signal recognition code - @Almeida-a @manuelgitgomes;
- [ ] Clean ROS launch files - @manuelgitgomes;

brnaguiar, 2022-05-15:
datasets:
    real/:
        [estadio/fnr/...]-[dia]-[mes]-[hora]/
        exemplo: estadio-02-01-12h30/
    gazebo/:
        gazebo-[dia]-[mes]-[hora]
        exemplo: gazebo-02-01-12h30/

modelos (.h5):
    [model]-[dia]-[mes]-[hora]-[loss].h5
    exemplo: rota-02-01-12h50-011.h5

manuelgitgomes, 2022-05-15:
As seen in other repos such as Major Alvega's and AtlasCar2's, a refactoring of our packages is needed to reduce the number of packages and the complexity.
I suggest doing as follows:

- prometheus_bringup: Every bringup file should be here, from writing, to driving, etc... This would be a junction of current physical_bringup, robot_driving and parts of simulation_bringup;
- prometheus_description: The same as the current traxxas_description;
- prometheus_driving: Everything used for the AD should be placed here (CNN, RNN, IPM), etc;
- prometheus_gazebo: Everything gazebo related should be here (arena, signal panel control, etc...). This would allow the various scripts and launch files of other packages to be the same for physical and simulation;
- prometheus_signal_recognition: The same as the current signal_recognition;

In the future, we can add other packages such as prometheus_vertical_signals, prometheus_parking, etc....

Examples shown here:
[image]
[image]

brnaguiar, 2022-05-15:
```
model:
    name: "..."
    developer: $automec_dev
    date: 25-01-2022 12h40
    ml_arch: 
        name: 'rota'
        epochs: 20
        batch_size: 30
    dataset: 
          name: "..."
          cam_pose: "..."
          developer: $automec_dev
          environment: ema
          frequency: 30
          image_size: 320,160
          linear_velocity: 0
          type: parking/driving/etc
          date:  25-01-2022 12h20
          comments: "..."
    model_eval: 5*
    comments: "..."
```

```
dataset:
  name: "..."
  cam_pose: "..."
  developer: $automec_dev
  environment: ema
  frequency: 30
  image_size: 320,160
  linear_velocity: 0
  type: parking/driving/etc
  date:  25-01-2022 12h20
  comments: "..."
```

manuelgitgomes, 2022-05-22:
Prometheus gazebo done.
Now for the prometheus_description, I think we should take the change and fix the urdf once and for all.

Almeida-a, 2022-05-23:
Signal Recognition
The code itself looks good. I have only carried out some detailed tweaks.
On a second task, we/I can also review the concepts of this module and reenforce the documentation, if you think is appropriate.

Also, I suggest we adding type hinting for better readability for the functions' signature, like the next example:
```python3
   def fun(value: int, matrix: numpy.ndarray) -> str:
      pass
```

manuelgitgomes, 2022-05-23:
Further changes for yaml:

- Change sigkill to pynput.keyboard;

Further suggestions:
- Add pynput,keyboard to drive.py to stop the program.
- Implement auto dataset naming;

manuelgitgomes, 2022-05-23:
Further changes in prometheus_gazebo:

- Add inverted chess to signal_panel;

Further suggestions:

- Add bringup for all sensors in physical world.

manuelgitgomes, 2022-05-23:
> Also, I suggest we adding type hinting for better readability for the functions' signature, like the next example:

We can also try docstrings! We can also add them for scripts.
https://docs.python.org/3/library/doctest.html#module-doctest
https://peps.python.org/pep-0257/

manuelgitgomes, 2022-05-23:
We should use:

- snake_case for variables;
- camelCase for functions;
- CamelCase for classes;

manuelgitgomes, 2022-05-23:
We should replace all global variables using partials and iterable variables.
Example:
https://github.com/lardemua/atom/blob/c3605c6c370eac286bc1d4163095d6e6685ae0e2/atom_calibration/scripts/dataset_playback#L213-L216

https://github.com/lardemua/atom/blob/c3605c6c370eac286bc1d4163095d6e6685ae0e2/atom_calibration/src/atom_calibration/dataset_playback/depth_manual_labeling.py#L39-L81

manuelgitgomes, 2022-05-23:
Use PEP-8.

Study walrus operator:

https://www.geeksforgeeks.org/walrus-operator-in-python-3-8/

```python3
sample_data = [
	{"userId": 1, "name": "rahul", "completed": False},
	{"userId": 1, "name": "rohit", "completed": False},
	{"userId": 1, "name": "ram", "completed": False},
	{"userId": 1, "name": "ravan", "completed": True}
]

print("With Python 3.8 Walrus Operator:")
for entry in sample_data:
	if name := entry.get("name"):
		print(f'Found name: "{name}"')

print("Without Walrus operator:")
for entry in sample_data:
	name = entry.get("name")
	if name:
		print(f'Found name: "{name}"')

```

manuelgitgomes, 2022-05-30:
Simulation is basically done. Sensors need to be added, meshes need to be changed and measures need to be taken. But our boy is here:
[image]

brnaguiar, 2022-05-30:
TASKS DONE:
- automatic dataset naming generation when performing data acquisition 
- pressing 'q' to quit gazebo and save the dataset

NEEDS TO BE DONE:
- add comment input

manuelgitgomes, 2022-05-30:
Change signals from 0º to 25º

brnaguiar, 2022-05-30:
bug: image capture starts automatically when running data acquisition command.

brnaguiar, 2022-06-05:
> * add comment input

Fixed

> bug: image capture starts automatically when running data acquisition command.

Done: it wasn't really a bug, it behaves like it should: only recording when in movement. 

I had an additional idea: sometimes, when performing the data acquisition task, we don't want to save the datasets due to some error, for example. In that case, I think we should implement an option to delete the created folder for the dataset and IMGs, using for example the keys: **q[uit]** and **s[ave]**, to discard the dataset and to save the dataset.

manuelgitgomes, 2022-06-13:
@Almeida-a try using a config dictionary for begin image, mask_mode, other config variables...

Almeida-a, 2022-07-25:
Made some clean-up [here](a8771700b15cf6441eba1e390c12df18908e0964)

manuelgitgomes, 2022-07-25:
Hello! Simulation is totally functional, using the command:

```bash
roslaunch prometheus_gazebo arena.launch
```

to launch gazebo, and the command

```bash
roslaunch prometheus_bringup bringup.launch sim:=true
```

to bringup prometheus inside gazebo. This launch file is also intended to be used in a real environment, changing the `sim` argument. Another argument can be used:

```bash
roslaunch prometheus_bringup bringup.launch sim:=true visualization:=true
```
where an rviz window displays the sensor data.

The bringup of the real robot is not yet implemented, but work has started on it.

A caveat, as the models changed folders, the user need to input this to their .bashrc/.zshrc:

```bash
export GAZEBO_MODEL_PATH="`rospack find prometheus_gazebo`/models:${GAZEBO_MODEL_PATH}" 
```

Almeida-a, 2022-07-25:
Task: delete the unpacking tokens that I introduced in this [commit](a8771700b15cf6441eba1e390c12df18908e0964).

Almeida-a, 2022-08-01:
> Task: delete the unpacking tokens (`**`) that I introduced in this [commit](a8771700b15cf6441eba1e390c12df18908e0964).

Done

manuelgitgomes, 2022-08-08:
Updates! Kinects are fully functional and compatibility between environments is achieved.

Dataset writing is also functional in simulation. Few tweaks need to be made to clean code and enhance compatibility with the real environment. One of these tweaks is the removal of androind_inputs to favor an automatically capped twist message (/cmd_vel).

Some "core" things were changed for compatibility purposes, but they can be revoked if chosen. cam_pose is removed from the yaml file in favor of only the urdf. If after every change the robot is calibrated, a new urdf is generated and the cam_pose is deprecated information. If there is no time to calibrate the robot, these changes can be stated on the comments section,

Known issues:
Simulation's pointclouds are dislocated, cannot find a way to straighten them.

Almeida-a, 2022-08-08:
> As refered in #155, pandas gives the following warning:
> 
> ```shell
> FutureWarning: The frame.append method is deprecated and will be removed from pandas in a future version. Use pandas.concat instead.
>   driving_log = driving_log.append(row, ignore_index=True)
> ```

Follow this [example](https://github.com/Almeida-a/ic-encoders-eval/blob/d9aa63ef4b98e20b83d49a6d67f09256f31b605b/procedure.py#L472-L476) to fix it:
```python
# Append new row to dataframe "stats"
row = pd.DataFrame(
    dict(filename=[outfile_name], cs=[cs], ds=[ds], cr=[cr], mse=[mse], psnr=[psnr], ssim=[ssim])
)
stats = pd.concat([stats, row])
```

manuelgitgomes, 2022-08-10:
Hello!
> Few tweaks need to be made to clean code and enhance compatibility with the real environment. One of these tweaks is the removal of androind_inputs to favor an automatically capped twist message (/cmd_vel).

This is now done, using the twist_cap.py script, in the prometheus_bringup package.
Controller support is added with joy_teleop.py in the prometheus_bringup package.
Some changes to the jupyter file were added, in order to automatize the model naming - @brnaguiar can you check if everything is good to you? The changes were in block 68.

manuelgitgomes, 2022-08-22:
Changes signal recognition to be functional in this new system. One large change, the signal recognition node now publishes the signal detected to the topic, leaving the decision to the driving node..
Further changes for this recognition:
- Changing from RGB to HSV (#158)
- Changing the sequential detection. The way it is, the system only defines a detection if detected on the last two frame. It can be more robust, such as in 4 of the last 5 frames.

manuelgitgomes, 2022-09-05:
> * Changing the sequential detection. The way it is, the system only defines a detection if detected on the last two frame. It can be more robust, such as in 4 of the last 5 frames

This is now implemented.

Autonomous driving in now reformulated. It reads the signal topic and acts accordingly (goes forward when 'pForward', stops when 'pStop', stops and end the program when 'pChess'). At the end the user is prompted to add comments and evaluate. 

With this, the reformation of the code is over. After testing with the real car, documentation is needed and then a merge to the main branch accompanied with release is my suggestion.
The release can follow something as @Almeida-a has done:
https://github.com/Almeida-a/imodec-dicoogle-plugin/releases/tag/v0.1.0-a0

manuelgitgomes, 2022-11-14:
Take pictures of setting up real car.

PedroMS3, 2023-06-08:
Everything should be done regarding this issue

closed 2023-06-08 by PedroMS3