AutoMecUA/AutoMec-AD #148: New power system for prometheus

manuelgitgomes, 2022-06-13:
Study and develop a new power system for prometheus using 2 LiPo 2S batteries.

andrefdre, 2022-11-20:
@manuelgitgomes is this issue related with what we were discussing last week?

manuelgitgomes, 2022-11-20:
Exactly!

andrefdre, 2022-11-20:
Okay, so to sum up what we discussed we thought about changing how the cameras will be powered, putting all the electronics in the same PCB and power the cameras with the same battery as the PWM signal generator. The other battery would power the motor.

andrefdre, 2022-11-23:
We bought 1 step-down and 1 step-up since the ones we had weren't working, making it possible to have now 2 cameras working at the same time.
We also changed how the power is distributed, having 1 battery powering all the logical electronics(PWM generator and the 2 cameras) and another battery powering the motor. This will fix one problem we had in the competition, which was the battery depleting too quickly.
The ESP32 will still be powered by the computer and not the battery since it needs the usb cable for data transmission.
For this we made a new PCB that will have all the components and do the power distribution. It has 4 screw connectors to plug the camera wires and another 2 for the battery. This need correct labeling to prevent mistakes.
The new PCB can be seen in the picture below:
[image]
[image]

andrefdre, 2022-11-23:
A new task that is needed is to create a schematic of the new PCB so it is well documented and easy to do maintenance in the future

andrefdre, 2022-12-12:
Schematic of the PCB board:
[Schematic.pdf](https://github.com/AutoMecUA/AutoMec-AD/files/10212419/Schematic.pdf)
[image]

manuelgitgomes, 2023-04-10:
This is already implemented

closed 2023-04-10 by manuelgitgomes