AutoMecUA/AutoMec-AD #149: Create development guides

Almeida-a, 2022-06-13:
Since we want all newcomers to integrate quickly and easily, I propose creating some software development guides, for example on the signal recognition package.

I will further detail this proposal shortly, like short, medium, and long-term goals.

In the meanwhile, any ideas are welcome!

Almeida-a, 2022-06-13:
We can take advantage of the generated docs (from the in-code docs as well as from the _.md_ files) and create a webpage using github pages.

Almeida-a, 2022-06-17:
Proposal for objectives:
 * Create a CONTRIBUTING.md describing guidelines for developers who would like to contribute to the project(s), whether on or off the university team.
 * Complete any unfinished in-code documentation
 * Create _.md_ files describing the techniques used to solve any of the main subjects' problems
 * Generate a (perhaps github) website using all of the produced documentation

I will organize this list into a short/medium/long term goals one.

PedroMS3, 2023-06-08:
some of the shores in this issue dont make sense anymore, but Completing any unfinished in-code documentation will be implemented in issue #201

closed 2023-06-08 by PedroMS3