AutoMecUA/AutoMec-AD #153: Create fast fastening system for car board

manuelgitgomes, 2022-07-25:
Create an intuitive and easy to use system for fixing the mdf board to the car.
[image]

manuelgitgomes, 2022-07-25:
@andrefdre is search for threads "porca de inserção" to use in this part

andrefdre, 2022-08-01:
After searching for 3D imbedded threads this was the best option:
https://www.amazon.es/-/pt/dp/B09NPYX4YH/ref=sr_1_3?crid=1Y715PH5CMWBS&keywords=tuerca+de+inserto+3d+m8&qid=1659384528&sprefix=porca+de+inser%C3%A7%C3%A3o+3d+m8%2Caps%2C82&sr=8-3

PedroMS3, 2022-08-21:
im not sure about the design of the insert for the imbedded threads on the part i made i think the threads wont stay in place as soon we torque the bolts

PedroMS3, 2022-08-22:
the design has been completed, just missing the printing fase

closed 2022-08-22 by PedroMS3