AutoMecUA/AutoMec-AD #155: Update python dependencies

Almeida-a, 2022-08-01:
I think we should update our python dependencies.
It is a good practice that can speed up our code.

We need to keep in mind, however, that there might be some breaking changes when doing this update.

manuelgitgomes, 2022-08-01:
Pay special attention to pandas, it gives warnings for a possible break.

Almeida-a, 2022-11-28:
The branch is `imp/update-dependencies`.

The versions were updated, and now a test run is due, in order to check the stability of the project.

@manuelgitgomes maybe you could help me with this?

manuelgitgomes, 2022-11-29:
Requirements updated with no breaking errors with the exception of pyyaml, which needs a older version due to ROS dependent code.
Now only "priting" errors occurs. Tensorflow prints messages constantly:
```
1/1 [==============================] - 0s 44ms/step
```

And this code:
https://github.com/AutoMecUA/AutoMec-AD/blob/03697e6a4923d36ca78a1a86b59984ff58e9d4dc/prometheus_decision/scripts/decision_maker.py#L137-L139
Is constantly printing.

@Almeida-a will figure this out. When this is done, we can possibly merge branch new/issue154 to dev. Do you approve @TatianaResend?

Almeida-a, 2022-12-05:
Regarding the tensorflow prints, and assuming the module that is responsible for those prints is `prometheus_driving/scripts/ml_driving.py`, the only tensorflow function called is the following:
https://github.com/AutoMecUA/AutoMec-AD/blob/ca64327f323b9ee22103e0a6b0ae395c298c25e0/prometheus_driving/scripts/ml_driving.py#L15

This means the [load_model](https://www.tensorflow.org/api_docs/python/tf/keras/models/load_model#args_1) function is the one generating those prints. However, there is nothing indicated in the docs about those prints, so the initial assumption might be incorrect.

manuelgitgomes, 2022-12-05:
```
2022-12-05 22:05:15.563763: I tensorflow/core/platform/cpu_feature_guard.cc:193] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2022-12-05 22:05:15.864661: W tensorflow/compiler/xla/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /home/nel/.local/lib/python3.8/site-packages/cv2/../../lib64:/home/nel/catkin_ws/devel/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
2022-12-05 22:05:15.864697: I tensorflow/compiler/xla/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
2022-12-05 22:05:16.787364: W tensorflow/compiler/xla/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libnvinfer.so.7'; dlerror: libnvinfer.so.7: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /home/nel/.local/lib/python3.8/site-packages/cv2/../../lib64:/home/nel/catkin_ws/devel/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
2022-12-05 22:05:16.787453: W tensorflow/compiler/xla/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libnvinfer_plugin.so.7'; dlerror: libnvinfer_plugin.so.7: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /home/nel/.local/lib/python3.8/site-packages/cv2/../../lib64:/home/nel/catkin_ws/devel/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
2022-12-05 22:05:16.787475: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Cannot dlopen some TensorRT libraries. If you would like to use Nvidia GPU with TensorRT, please make sure the missing libraries mentioned above are installed properly.
[INFO] [1670277918.247413]: Using model: /home/nel/catkin_ws/src/AutoMec-AD/prometheus_driving/scripts/../models/rota-29-11-2022-19h31m-0.01.h5
2022-12-05 22:05:18.295710: E tensorflow/compiler/xla/stream_executor/cuda/cuda_driver.cc:267] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2022-12-05 22:05:18.295751: I tensorflow/compiler/xla/stream_executor/cuda/cuda_diagnostics.cc:156] kernel driver does not appear to be running on this host (shedinja): /proc/driver/nvidia/version does not exist
2022-12-05 22:05:18.296470: I tensorflow/core/platform/cpu_feature_guard.cc:193] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
1/1 [==============================] - 0s 173ms/step

```

manuelgitgomes, 2023-01-02:
Obnoxious prints removed. Issue closed.

commit a5559701 (2023-01-02): Closes #155 - ends undesirable prints.
commit d5ad3725 (2023-01-02): Imp/requirements (#177)

* #155 Update requirements with minor printing errors

* Closes #155 - ends undesirable prints.
closed 2023-01-02 by manuelgitgomes