AutoMecUA/AutoMec-AD #168: Implement external PWM generator with esp32

andrefdre, 2022-11-14:
Previously was implemented an external PWM generator but wasn't working after some testing we figure out that it was being supplied with 3.3V and the motor was expecting 5V. With this we decided to create a new PCB with the new esp32 that will also be useful with the encoder due to it's 2 cpus and the PCA9685 that will generate the PWM signals.

inaciose, 2022-11-15:
I told to @manuelgitgomes that maybe the problem. 
Before make a new pcb we should improve that one and test if it drives the esc with the 5 volts.

andrefdre, 2022-11-15:
We talked about doing a new PCB  since the esp8266 doesn't have a 5V.

inaciose, 2022-11-15:
show me a good picture of the pcb please. As far as I remember I belive we may change it to have 5 volts.

andrefdre, 2022-11-15:
There is one in issue #141

inaciose, 2022-11-16:
Please check the ESP8266-Pinout

https://lastminuteengineers.b-cdn.net/wp-content/uploads/iot/ESP8266-Pinout-NodeMCU.png

The vin pin is 5V

I found this photo and it shows some connections. If it powered by usb you need to check if it have 5 volts in Vin pin, it it have we need to change the connection to pca9685 vcc to the vin pin insted the 3.3V pin.
It it doesnt have 5 volts we need to get 5 volts from another source, suply it to the esp in vin pin and use the same source to connect to the pca9685 vcc.

[image]

andrefdre, 2022-11-20:
The board being powered with 3.3V was the problem. I changed to 5V and it fixed it. Also we Implmented the board with the esp32 and changed how the cameras will be powered to place all the electronics of the car in the same PCB.

closed 2022-11-21 by manuelgitgomes

andrefdre, 2022-11-23:
Sometimes the motor worked and sometimes it didn't this was due to the frequency was't being 50 Hz exactly so we adjusted this through the help of the oscilloscope and now it works. We also tested the motor PWM duty cycle length and we found that the completely backwards is 1000 ms and completely  forwards is 2000 ms and idle is around 1500 ms.