AutoMecUA/AutoMec-AD #172: Detect traffic signal base

manuelgitgomes, 2022-11-21:
Points to account for:

- [ ] Fix vertical signals/traffic cones not appearing in gazebo;
- [ ] Alter pose of bottom kinect;
- [ ] RGBD detection signal base.

brunofavs, 2022-11-23:
Tried to implement it with a more generic elegant way but wasn't succeded

for d in "`rospack find prometheus_gazebo`"/models; do export GAZEBO_MODEL_PATH="$d:${GAZEBO_MODEL_PATH}";

brunofavs, 2022-11-23:
> Added the following lines to .zhsrc to include the vertical signals again.
> 
> This isn't well formatted, im having trouble with the code special characters
> 
> ```
> # Vertical signals
> ```
> 
> export GAZEBO_MODEL_PATH="`rospack find prometheus_gazebo`/models/VerticalSignals/circular_signals:${GAZEBO_MODEL_PATH}" export GAZEBO_MODEL_PATH="`rospack find prometheus_gazebo`/models/VerticalSignals/quadrangular_signals:${GAZEBO_MODEL_PATH}" export GAZEBO_MODEL_PATH="`rospack find prometheus_gazebo`/models/VerticalSignals/triangular_signals:${GAZEBO_MODEL_PATH}"

Also im yet to update the wiki/readMe to include this new information

manuelgitgomes, 2023-03-29:
@Ze6000, please add a subscriber for depth camera, similar to the one already presented for rgb:
https://github.com/AutoMecUA/AutoMec-AD/blob/ad93b4a479fb0e62df563d105be890bd69f1d363/prometheus_signal_recognition/scripts/signal_foot_recongnition.py#L18-L31