AutoMecUA/AutoMec-AD #175: Build a new PCB  box and 2 battery shields to put under the car's board

PedroMS3, 2022-12-04:
the necessity to  move the initial location of the batteries and PCB box to under the car's board was identified, it creates more space on top of the board

PedroMS3, 2023-01-23:
[image]
this is the new battery holder (it's missing another one on the other side of the body)

PedroMS3, 2023-01-23:
it was suggested to use the body holding the battery holders to also hold the PCB, might use some "C" profiles on the sides of the PCB

PedroMS3, 2023-01-25:
New design that already includes a part to position the PCB
[image]
[image]
Just missing the confirmation of some measurements

manuelgitgomes, 2023-01-25:
Nice job, looks great! We can take these measurements next week!

manuelgitgomes, 2023-02-08:
The PCB has 3mm of thickness and 80mm of width

PedroMS3, 2023-02-09:
can we rotate the PCB 90º??
It would be easier for the design of the hardware

andrefdre, 2023-02-09:
Yes, I think we can the only restriction would be the battery cable, but I think it has more than enough.

PedroMS3, 2023-02-20:
I am going to prepare the files for printing and testing

manuelgitgomes, 2023-03-06:
Create a new part for step-down

PedroMS3, 2023-03-06:
[image]

the part for the step-down is done, the stl is on the hardware chat on Teams

PedroMS3, 2023-03-13:
[image]
the technical drawing has been made, i wil try to take the machined part to the next meeting

PedroMS3, 2023-03-20:
the machined part is ready, i just have to re-drill some holes on the printed part

[image]

PedroMS3, 2023-04-05:
some changes need to be made in the fixtures of the PCB

PedroMS3, 2023-04-12:
the "aeroplano" part has been mounted in the car chassis. It's missing the two "asa" pieces
[image]

manuelgitgomes, 2023-04-17:
[image]

Printed battery supports

PedroMS3, 2023-05-08:
the supports for the pcb have to be redesigned and reprinted

PedroMS3, 2023-05-22:
the supports have been sucessfully mounted!! thanks to francisco freitas they were printed
[image]

closed 2023-05-22 by PedroMS3