AutoMecUA/AutoMec-AD #192: Adapt LSTM for velocity/direction

manuelgitgomes, 2023-04-17:
Adapt current LSTM code to also predict velocity control, alongside direction.

The code is on the `testing_cnn_architectures`, we should make a pull request to `dev` soon.

To learn pytorch, use [this video](https://www.youtube.com/watch?v=c36lUUr864M) as a tutorial.

brunofavs, 2023-07-31:
A note for the next meeting.
In the teams files, the models folder could be renamed to trained models to reduce confusion.

-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//-------//

So far I'm studying the scripts and how they interact with each other.

- I figured out I need to record a new dataset with varying velocity because the ones we have all have constant values.
- I need to incorporate the folder structure and alias for the scripts to recognize the datasets and trained models.
- Next steps are also figuring out how to record a new dataset and how to measure the current model's performance to have a way of benchmarking

brunofavs, 2023-08-02:
Today's progresses:

- Was having some issues running a previously trained model, the problem was just trying to run on an old architecture still on dev.
-Created a new branch from andre's fixed lstm and tried to train a new model on a old dataset and went into some issues because I didn't know I had to run a statistics script and use a certain cropsize and a low batchsize.

- After debugging and finally training a new model, tried running and it worked very nicely. Up to velocities of 1.5 the model still held good.

Im wondering how I will compare the new model with the new input later on to the current one.

For tomorrow I want to train a new dataset with varying velocities, start figuring out how to change the architecture and see from there.
I should also not forget this line [in model_train.py](https://github.com/AutoMecUA/AutoMec-AD/blob/5b642cc5e3fc4d33d871dc13be9ddaa8f85b8941/prometheus_driving/scripts/model_train.py#L91)

brunofavs, 2023-08-03:
Today's progresses :

- Implemented a argument to include variable velocity in the bringup.
- Implemented the reverse trigger in the controller similar to the forward trigger. It also uses the entire throw of the trigger instead of half like before to be easier to drive at lower velocities
- Recorded a dataset with variable velocity
- Trained a 30epoch lstm model with the variable velocity dataset, still only inputting the images into the network. It did not work well, even though the dataset was 11k images

brunofavs, 2023-08-05:
Today's progresses:

- Updated the architecture to take velocity as input
- Trained a model on the new architecture 
- Tested the new model

The dataset might have been recorded without taking some precautions to ensure the quality of the data. The car drives very nicely up to 50% faster than the older velocity yet when it gets to a couple of specific spots on the track just wanders off completely.