AutoMecUA/AutoMec-AD #197: Create low battery alert

manuelgitgomes, 2023-05-08:
Create an electric system that notifies the user that the car is running low on battery:

- Use buzzers, LEDs, or another notification system
- Explore the usage of balance connector to read the voltage in each battery
- Differentiate the signal for each battery

Carolf27, 2023-06-26:
Our ideia is to read the batteries' voltage and when it reaches a minimum level an alarm is sent off and we know we need to change it and recharge it

Carolf27, 2023-06-26:
Last wednesday (june 21st) we tested reading the voltage from an unused battery with an old esp.

We conected the pins from the balance connector to the esp and were able to read the voltage.

We first measure the battery's voltage with the reader we use and that battery had around 7.8 V, and when we read its voltage the result was 1024 meaning that the voltage that was reaching ESP was the highest, we concluded that this was happening because the esp can only read up to 3.3V or 5V, so we had to reduce the voltage.

So then we used two resistences as an tension devider (120 ohm and 220 ohm) and we got good results because now the esp was givin values below 1024 meaning that it was reading the battery's voltage.

Now we want to explore using an opamp insted of a tension devider.