AutoMecUA/AutoMec-AD #201: Improve Wiki documentation

andrefdre, 2023-05-28:
General improvement to the wiki.

Adding controller installation step-by-step procedure.

andrefdre, 2023-06-05:
I found out why the phone wasn't able to communicate with the car. Basically, the images in the wiki explain to set the joystick topic to android_input_dir but for it to work it should be cmd_vel and the button for velocity isn't used anymore.

andrefdre, 2023-06-08:
Documentation to all the scripts should be implemented, adding description to all the functions and a general description of what the main code is for similar to:

"""
    Script for driving the car using a Deep-learning model.
    The model is loaded from a .pkl file and the image is obtained from a ROS topic.
    The model is loaded from the path specified in the parameter 'model_name'.
    The image is obtained from the topic specified in the parameter 'image_raw_topic'.
    The steering angle is published in the topic specified in the parameter 'model_steering_topic'.
    The model can be changed by publishing the name of the model in the topic '/set_model'.
"""

Ze6000, 2023-06-19:
The users' guide for software installation needs to be tested on a computer recently installed with Linux to ensure its accuracy and correctness.
The simulation and operation of the real car was successful.
There is a bug encountered during the recording of the dataset in the real car.
The code provided after recording the dataset has not yet undergone testing.

PedroMS3, 2023-07-03:
the wiki has been tested and aproved

closed 2023-07-03 by PedroMS3