AutoMecUA/AutoMec-AD #205: Purchase the new camera and sensors

Ze6000, 2023-10-04:


PedroMaia21, 2023-10-11:
[image]

This schematic represents an idea to the sensors positioning. ECU represents the place of the processing unit, IMU represents the placemento of the 6DoF sensor, Radar is the placement of ultrassonic sensors and lidar represents the placement of the laser sensors. The lidar on the hood could be a rotatory one, but is something yet to be studied if its really worth it

PedroMaia21, 2023-10-14:
Here are some links on what is the expected component:
-6DoF sensor: **MPU6050** - [AliExpress](https://pt.aliexpress.com/item/1005004160442370.html?src=google&src=google&albch=shopping&acnt=631-313-3945&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbActive=false&isSmbAutoCall=false&needSmbHouyi=false&albcp=19373920267&albag=&trgt=&crea=pt1005004160442370&netw=x&device=c&albpg=&albpd=pt1005004160442370&gclid=CjwKCAjw-KipBhBtEiwAWjgwrFHH3y_DrZYfJcp6U5nb-AMlxcSQ_AXqo4XsUKrHLNe--KJpJKjfzRoC0RMQAvD_BwE&gclsrc=aw.ds&aff_fcid=21250317489644fda59f42d7d22262ce-1697292641622-04552-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=21250317489644fda59f42d7d22262ce-1697292641622-04552-UneMJZVf&terminal_id=c10ef5969a314df5856a319b913cbd33&afSmartRedirect=y)
-Radar: **HC-SR04** - [Mauser](https://mauser.pt/catalog/product_info.php?products_id=096-6220&utm_source=google&utm_medium=cpc&utm_campaign=shopping_catalog_pt_011&utm_content=feed&gclid=CjwKCAjw-KipBhBtEiwAWjgwrBRJnbGOJeMBe1w-XicYpBJkT6vczbPsuBMOGiOf-PNKYfLmzHqqJxoC9C8QAvD_BwE)
-Laser: **TF-Luna** - [Mauser](https://mauser.pt/catalog/product_info.php?products_id=095-0752&utm_source=google&utm_medium=cpc&utm_campaign=shopping_catalog_pt_011&utm_content=feed&gclid=CjwKCAjw-KipBhBtEiwAWjgwrH9T6iVUQLUZDwDcDY0TA00ZqmppWPoMuh4cuVX0VMvErCrXnnx9lhoCVPoQAvD_BwE)

Hugo-Sobreiro, 2023-10-25:
camera we are considering: [intel](https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d435i.html?_ga=2.204936211.287287017.1698247348-253607904.1697036177)

PedroMaia21, 2023-11-15:
the laser sensors were updated: [laser](https://mauser.pt/catalog/product_info.php?cPath=1667_2669_2675&products_id=096-8830)

Hugo-Sobreiro, 2023-11-20:
lower price camera we are considering: 
[Orbbec Astra S](https://www.ebay.com/itm/295920788449?norover=1&mkevt=1&mkrid=711-167022-160074-6&mkcid=2&itemid=295920788449&targetid=295818637131&device=c&mktype=pla&googleloc=1011706&poi=&campaignid=20741944927&mkgroupid=158218881067&rlsatarget=pla-295818637131&abcId=&merchantid=119648210&gclid=Cj0KCQiApOyqBhDlARIsAGfnyMqUSCBehP6N-UKDLjRCyeskJ5h83OuogTH-Gb7XfvUsvoZcLj2d_icaAno4EALw_wcB) -> [specs](https://shop.orbbec3d.com/Astra-S)
[GEMINI E WITH PACKAGING/ASTRA STEREO S U3](https://pt.aliexpress.com/item/1005004862291108.html?src=google&src=google&albch=shopping&acnt=272-267-0231&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=pt1005004862291108&ds_e_product_merchant_id=107792690&ds_e_product_country=PT&ds_e_product_language=pt&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20695431540&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=4&gclid=Cj0KCQiApOyqBhDlARIsAGfnyMpsBYAdOzs2KP1JBD3zzzS9Oz2-rixwOTd6gZNC3vTj9dZQ0uBaOokaAiV5EALw_wcB&aff_fcid=3974979805dd4464abcee4ea0a9e81bb-1700517598079-07473-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=3974979805dd4464abcee4ea0a9e81bb-1700517598079-07473-UneMJZVf&terminal_id=d2393a5d5ddc4dd7b9f3d377782dca26&afSmartRedirect=y) -> [spec1](https://shop.orbbec3d.com/product/532) [spec2](https://shop.orbbec3d.com/Orbbec-Astra-Stereo-S-U3)

PedroMS3, 2024-04-29:
JPS will buy us the following equipment :

- IM130918002 - MPU6050 Module Triple-Axis Gyroscope
- Cabo HDMI para DVI 5m
- Cabo Rede Flat UTP6 20Metros Cinza
- Cabo USB-A 2.0 2A Macho / Micro USB-B Macho Preto 1m
- 2x baterias LiPo 11.1V,  7100mAh
- 4x Módulo Encoder rotativo compatível com Arduino - HW-040
- 3x Módulo de câmara V2 oficial Raspberry Pi - 8MP - Raspberry Pi
- 10x Sensor ultrassónico (HC-SR04) compatível com Arduino