AutoMecUA/AutoMec-AD #206: Find and purchase a Nvidia suitable for the new car

Ze6000, 2023-10-04:


andrefdre, 2023-10-16:
Resposta do email que enviei:

Estive agora mesmo a falar com um colega teu sobre este assunto. Já trabalhei com um Nvidea Jetson Nano. Este nao tem Bluetooth nem wifi (tem entrada ethernet, para ligar à rede). No entanto, consegues arranjar um modulo bluetooth para usar na placa, com interface USB que seja (em termos de energia consome mais do que o BLE do ESP32, por exemplo). Contudo tens capacidade computacional para correr e treinar algoritmos e consegues ter um OS baseado em Linux.

Existem placas da Nvidea com maior poder computacional (Jetson Xavier por exemplo), mas depende também do orçamento que tenham disponível. Em geral, em termos computacionais, o Nvidea Jetson Nano dá conta do recado. Tenham em conta se a placa vem ou nao com memória emmc. É mais fiavel e rápido usar essa memória do que um cartao SD.

A titulo de curiosidade, ve este artigo, onde é feita comparaçao entre várias placas (https://www.sciencedirect.com/science/article/pii/S1568494620309911).

andrefdre, 2023-10-16:
I read the paper lightly the other day, and it didn't really seam to have much information, but give me your thoughts.

PedroMS3, 2023-10-22:
There is a fellow ex-Automec AD member, Bruno Monteiro, that has a board similar to the ones we are looking at he said he could lend us the board so that we could start experimenting on it.  
[image]
[image]

andrefdre, 2023-10-22:
Nice, the first thing we should test is if it has capacity to run ROS and both neural networks at the same time.

manuelgitgomes, 2023-10-23:
Hello @andrefdre!

As a last option, the models can be converted from torch to TensorRT, a framework by Nvidia that supposed makes models lighter on inference.
I have never done this myself, but know some people that did and I know it is a arduous process, so view it as a last option.

Have some links that might help you:
https://github.com/NVIDIA-AI-IOT/torch2trt
https://medium.com/@zergtant/accelerating-model-inference-with-tensorrt-tips-and-best-practices-for-pytorch-users-7cd4c30c97bc

andrefdre, 2023-11-13:
I tried to use the Jetson Nano but unfortunately didn't have much luck since it didn't come with a micro SD card and the only one I had was 64gb which eventually got out of space. During my tests, I found out Ubuntu isn't directly supported in the Jetson nano, but I found a repository that has Ubuntu for jetson Nano https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image. I dunno which one is better to have Ubuntu since it can have limitations I am not currently aware, or use docker using the Nvidia linux image.

When I tried the Ubuntu image, I manage to install ROS, but it took me a little more effort since it is not as straight forward due to missing dependencies. I couldn't install everything due to lack of space.

PedroMaia21, 2023-11-22:
Is this problem related with the Nvidia Jetson kit or with the storage in the sd card? Because my interpretation of the regulations makes me think that we dont have a lot of margin in the choice of the development kit

andrefdre, 2023-11-27:
> Is this problem related with the Nvidia Jetson kit or with the storage in the sd card? Because my interpretation of the regulations makes me think that we dont have a lot of margin in the choice of the development kit

It is related to the board but also the SD card because I ran out of space and couldn't install to test everything.

andrefdre, 2023-11-29:
I think first we should get a SD card with more space and test if we can make the code work like this. If not we have to think what's the best strategie and research more cause I don't know if other boards we won't have the same problem.

PedroMaia21, 2023-11-29:
Does a 128GB SD card have enough space?

PedroMaia21, 2023-11-29:
That seems a good startign point, thanks

PedroMaia21, 2023-11-29:
[image]
And this is the feedback on investing on a Xavier board, instead of the Nano

PedroMS3, 2023-11-30:
[image]
Maybe we could try and talk to the students in this email.

PedroMS3, 2023-12-11:
we should ask the ADC VDI organisation about any guidelines on this topic.

andrefdre, 2023-12-18:
I searched about Jetson xavier and found out that it can have ROS Melodic instead of Noetic, which is an older version. https://www.stereolabs.com/blog/ros-and-nvidia-jetson-xavier-nx

andrefdre, 2023-12-18:
We could try to contact Nvidia asking for their suggestion.