AutoMecUA/AutoMec-AD #207: Create a bluetooth interface to start the car

PedroMS3, 2023-10-09:
In the competition there is time frame where the car must start moving after a bluetooth signal is sent, coincident with the greenlight a luminous signal, the purpose of this issue is to create an bluetooth interface to accomplish this objective

andrefdre, 2023-10-16:
Code needed to run the Servo driver

```
#include <Wire.h>
#include <Adafruit_PWMServoDriver.h>

#define ESCMIN  1000 // This is the 'minimum' pulse length count of the ESC
#define ESCMAX  1960 // This is the 'maximum' pulse length count of the ESC
#define SERVOMIN  500 // This is the 'minimum' pulse length count of the steer Servo
#define SERVOMAX  2500 // This is the 'maximum' pulse length count of the steer SErvo
#define SERVO_FREQ 50 // Analog servos run at ~50 Hz updates

void setup() {
  // PCA9685 board
  pwm.begin(); // Begins comunnication
  pwm.setOscillatorFrequency(26315700); // This frequency can be tuned to adjust the generated pwm wave
  pwm.setPWMFreq(SERVO_FREQ);  // Analog servos run at ~50 Hz updates

}

void loop(){
// Convert from ms to angle between 0 and 180
  int escms = map(90, 0, 180, ESCMIN, ESCMAX);   // scale it to use it with the servo library (value between 0 and 180)
  pwm.writeMicroseconds(ESC, escms);  // Generates the PWM wave
}
```

Carolf27, 2023-10-23:
We tried to move the car forward with a phone via bluetooth with this code:

```
#include "BluetoothSerial.h"

#include <Wire.h>
#include <Adafruit_PWMServoDriver.h>

#define ESCMIN  1000 // This is the 'minimum' pulse length count of the ESC
#define ESCMAX  1960 // This is the 'maximum' pulse length count of the ESC
#define SERVO_FREQ 50 // Analog servos run at ~50 Hz updates
#define ESC 0 // Defines the chanel for the esc

//#define USE_PIN // Uncomment this to use PIN during pairing. The pin is specified on the line below
const char *pin = "1234"; // Change this to more secure PIN.
int value = 20;

String device_name = "ESP32-BT-Slave";

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;
String message = "";
char incomingChar;

// Initializes the pwm board generator
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

int potValue;  // value from the analog pin

void setup() {
  Serial.begin(115200);
  SerialBT.begin(device_name); //Bluetooth device name
  Serial.printf("The device with name \"%s\" is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str());
  //Serial.printf("The device with name \"%s\" and MAC address %s is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str(), SerialBT.getMacString()); // Use this after the MAC method is implemented
  #ifdef USE_PIN
    SerialBT.setPin(pin);
    Serial.println("Using PIN");
  #endif
 
   // Read received messages (LED control command)
  if (SerialBT.available()){
    char incomingChar = SerialBT.read();
    if (incomingChar != '\n'){
      message += String(incomingChar);
    }
    else{
      message = "";
    }
    Serial.write(incomingChar);  
  }

  // PCA9685 board
  pwm.begin(); // Begins comunnication
  pwm.setOscillatorFrequency(26315700); // This frequency can be tuned to adjust the generated pwm wave
  pwm.setPWMFreq(SERVO_FREQ);  // Analog servos run at ~50 Hz updates

}

void loop() {

 

   // Read received messages (LED control command)
  if (SerialBT.available()){
    char incomingChar = SerialBT.read();
    if (incomingChar != '\n'){
      message += String(incomingChar);
    }
    else{
      message = "";
    }
    Serial.write(incomingChar);  
  }

  // Check received message and control output accordingly
  if (message =="on"){
    Serial.println("ON");
     // Convert from angle to ms between 0 and 180 to the ms min and ms max
    value = 90;
    Serial.print(value);

  }
  else if (message =="off"){
    value = 0;
    Serial.println("OFF");
    Serial.print(value);

  }
  Serial.print(value);
  int escms = map(value, 0, 180, ESCMIN, ESCMAX);   // scale it to use it with the servo library (value between 0 and 180)
  pwm.writeMicroseconds(ESC, escms);  // Generates the PWM wave
  //delay(20);
}
```
Note: This code is not commented but @Carolf27 will do that when possible

Resultof using this code: The car responded to the "off" button.

Carolf27, 2023-11-08:
**How to run Serial Bluetootj terminal**
Initial page:
[image]

From the initial page go to the up left corner to open the menu. Click on Devices to conect to the Esp. 
[image]

The device name should appear. Click on the name and begin connection. In this case the name is "ESP32-BT-Slave".
[image]

Now you need to configure the buttons. We will use the M1 and M2. Make a long press on each one to open the button's configuration menu. Fill each one like this:
[image]
[image]

Now if you click on the buttons the message "on" and "off" will be sent to the esp so that the car moves forward or stop. You can also send messages to the arduino serial through the bar on the botton in the initial page.

Carolf27, 2023-11-15:
> Code needed to run the Servo driver
> 
> ```
> #include <Wire.h>
> #include <Adafruit_PWMServoDriver.h>
> 
> #define ESCMIN  1000 // This is the 'minimum' pulse length count of the ESC
> #define ESCMAX  1960 // This is the 'maximum' pulse length count of the ESC
> #define SERVOMIN  500 // This is the 'minimum' pulse length count of the steer Servo
> #define SERVOMAX  2500 // This is the 'maximum' pulse length count of the steer SErvo
> #define SERVO_FREQ 50 // Analog servos run at ~50 Hz updates
> 
> void setup() {
>   // PCA9685 board
>   pwm.begin(); // Begins comunnication
>   pwm.setOscillatorFrequency(26315700); // This frequency can be tuned to adjust the generated pwm wave
>   pwm.setPWMFreq(SERVO_FREQ);  // Analog servos run at ~50 Hz updates
> 
> }
> 
> void loop(){
> // Convert from angle to ms between 0 and 180 to the ms min and ms max
>   int escms = map(90, 0, 180, ESCMIN, ESCMAX);   // scale it to use it with the servo library (value between 0 and 180)
>   pwm.writeMicroseconds(ESC, escms);  // Generates the PWM wave
> }
> ```

@andrefdre from wich file did you get this code? Because now we can make the car go forward and backwards and we want to use the steering servo to make it go left and right and i don't know what is its channel

andrefdre, 2023-11-16:
I took that code from the code we have in the car. It's in the dev branch inside Prometheus_bringup