AutoMecUA/AutoMec-AD #209: Make a "raincoat" for the new car

PedroMS3, 2023-11-06:
Because of the bad weather the new car might get wet therefore a protection against the weather should be made.
As a plus a new transportation method like a trolley could be included.

PedroMS3, 2023-11-15:
I was talking to engenheiro Festas and he told me we could have a protective foam he has laying around the workshop, but we do have to go there and un-glue it from the wood pieces it is attached to.
[image]

FSFToco, 2023-11-20:
In the search for a cheap suitcase to later modify into the car raincoat, I found that the most promising one, being the cheapest I could find that is waterproof and big enough, costs 40€ [image] 
In the event that we need a bigger suitcase, the prices range from 40/60 when around that size.

PedroMS3, 2023-11-20:
Further research should be made on products that swell once they are aplied to the object. Search for "espuma poliuretano"

PedroMS3, 2023-11-29:
these were the possible sponsors we sent an email and the minimum lengths the suitcas should have.
[image]
[image]

FSFToco, 2023-12-06:
https://m.olx.pt/d/anuncio/mala-de-viagem-rigida-preta-IDGRaYq.html
possible choice

FSFToco, 2023-12-06:
https://m.olx.pt/d/anuncio/mala-rgida-samsonite-IDHxeZw.html another possible option

PedroMS3, 2023-12-13:
Samsonite's Answer :
[image]

PedroMS3, 2023-12-13:
[image]
Updates

PedroMS3, 2024-02-26:
According to discussed, a rough 3D model will be made to understand the layout of the sponge and the space needed.

FSFToco, 2024-03-06:
A rough model of the car was built, for the purpose of understanding its size and shape in relation to the bag (which we also obtained like a month ago from the previously shown contact).
[image]

FSFToco, 2024-03-06:
[image]
the side of the suitcase where the car will be placed was modeled in CAD

FSFToco, 2024-05-08:
[image]

Modeled the first sponge support of the raincoat project, this was made to house the box containing the batteries, as well as the box pertaining to the batterie's charger. Has a side indentation to make it a better fit against the side of the suitcase where the locking mechanism is.