AutoMecUA/AutoMec-AD #212: Create PCB

Carolf27, 2024-01-25:
Create PCB to connect the battery to the motor and the sensors and to nvidia

joaoafc, 2024-03-06:
link to PCB schematic
https://u.easyeda.com/join?type=project&key=faff1aad60f9933ea1d3cb5e0197abb9&inviter=ace481dd0702426a9fe833ea2ffc4ef5

joaoafc, 2024-03-06:
Material needed to build the pcb:
PCA9685
electronic board
Cables
Step-down 11.1V to 5V

PedroMS3, 2024-04-03:
There are a few things that need to be checked. How to control the car with the RC remote and with the Nvidia simultaneously, and to add a power supply for the transponder, like it is said in the rulebook.

Ricardof19, 2024-05-24:
[image]

[image]

[image]

Here is the initial theoretical diagram of the electrical components. We are still studying some possible changes.