AutoMecUA/AutoMec-AD #215: Sensor reading scripts

PedroMaia21, 2024-03-20:
The sensors to be added to the car are:

- Ultrassonic - [HC-SR04](https://mauser.pt/catalog/product_info.php?products_id=096-6220&utm_source=google&utm_medium=cpc&utm_campaign=shopping_catalog_pt_011&utm_content=feed&gad_source=1&gclid=Cj0KCQiA-62tBhDSARIsAO7twbagSz9An0i0pnkBv5op1zoUm4SScpNrdwGCuIvxFUGwlYmsJf51KskaAiNyEALw_wcB)7
- Sensor IMU - [MPU-6050](https://www.amazon.es/dp/B00LP25V1A/ref=asc_df_B00LP25V1A/?tag=ptgogshpadde-21&linkCode=df0&hvadid=634362297348&hvpos=&hvnetw=g&hvrand=4312067211137699805&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=20865&hvtargid=pla-407209664611&psc=1&language=pt_PT&mcid=2e6cf619c19437ceb28e49b291a36b7b)
- Encoder - [HW-040](https://mauser.pt/catalog/product_info.php?products_id=096-9086&utm_source=google&utm_medium=cpc&utm_campaign=shopping_catalog_pt_011&utm_content=feed&gad_source=1&gclid=Cj0KCQiA-62tBhDSARIsAO7twbZPlWnoq1lHPVrHaB7obJZSE4NPvBq8jOwp5YJ-EU6h4_dTWfkZxrUaAnjqEALw_wcB)
- LiDAR - [A1M8-R6](https://www.botnroll.com/pt/outros/4814-rplidar-a1m8-r6-slamtec-scanner-laser-360.html)
- Main Camera - [Orbbec Astra S](https://www.ebay.com/itm/295920788449?norover=1&mkevt=1&mkrid=711-167022-160074-6&mkcid=2&itemid=295920788449&targetid=295818637131&device=c&mktype=pla&googleloc=1011706&poi=&campaignid=20741944927&mkgroupid=158218881067&rlsatarget=pla-295818637131&abcId=&merchantid=119648210&gclid=Cj0KCQiApOyqBhDlARIsAGfnyMqUSCBehP6N-UKDLjRCyeskJ5h83OuogTH-Gb7XfvUsvoZcLj2d_icaAno4EALw_wcB)
- Secondary camera - [Raspberry Pi Camera Module 2](https://mauser.pt/catalog/product_info.php?products_id=096-4061&utm_source=google&utm_medium=cpc&utm_campaign=shopping_catalog_pt_011&utm_content=feed&gad_source=1&gclid=Cj0KCQiA-62tBhDSARIsAO7twbbeINIGhTC5iqmTpkA7lWvxPT6xl7Iwf5WzRAG5AKXuwhdqOGtwvPkaAqofEALw_wcB)

PedroMaia21, 2024-03-20:
The first step should be the research and planning of the scripts for the sensor reading

Rondeler, 2024-05-08:
the scripts for the ultrassonic sensor and the program for reading frames from the main camera are being finished

Rondeler, 2024-06-05:
the ultrassonic sensor script is finished and already uploaded into the jetnano. We tried to execute the program but failed. we think its because of a library issue