AutoMecUA/AutoMec-AD #216: Motor controlling scripts

PedroMaia21, 2024-03-20:
Should be created the scripts for controlling the motors and the direction servo.

Components:
- Motor - [FLUX EMH-3S ESC](https://www.hpiracing.com/en/part/112850)
- Controller - [FLUX MMH-4000KV](https://www.hpiracing.com/en/part/112851)
- Servo - [HPI SS-20WR SERVO](https://www.hpiracing.com/en/part/120018)

PedroMaia21, 2024-03-20:
The first step should be the research and planning of the scripts for the car controlling

Hugo-Sobreiro, 2024-04-17:
[image]

Motor connection diagram.

Hugo-Sobreiro, 2025-03-26:
We were able to control the motors using the ESP32. The servo motor works normally with the ESP32Servo.h library, but it has a physical limitation of about 30º to each side. So when set to 60º, it turns fully to the left, 90º is centered (forward), and 120º turns fully to the right.

As for the DC motor, we had a few issues getting the ESC to behave exactly as we wanted. We found that the motor speed is controlled by the PWM pulse width in microseconds — 1500 µs puts the motor in neutral, anything below that moves it in reverse, and above that drives it forward. There was also a delay caused by a built-in safety feature of the ESC: in order to switch from forward to reverse, the ESC requires a specific sequence. You need to first send a reverse PWM signal, then a neutral signal, and finally the reverse PWM again — only then will it actually allow the motor to move backward.

Here is the code we used with TCP/IP communication for the wireless control of the motors for future tests.

[CarroTCPIP.zip](https://github.com/user-attachments/files/19472669/CarroTCPIP.zip)